PaulStoffregen/USBHost_t36 #1: Bug when sending a lot of midi messages to teensy 3.6

Dealerpriest, 2017-08-21:
I've used this library to create a usb-midi host. It reads incoming noteOn messages from a keith McMillen BopPad and trigger sounds using the audio library (```drum.noteON()```).
It works well for the most of the time. But if I flood the teensy by playing a a lot on the drum pad, the teensy starts acting weird. Instead of separating between incoming messages it starts reading several noteOns when there is actually only one.
The drum pad sends note values 36, 38, 42 and 47. Sending several of these notes together by playing intensely on several pads at the same time results in the lock up described. 
I've found that it's possible to get out of the lock up by playing several notes simultaneously again.

gritd, 2017-12-07:
This is probably not a bug but I suspect the midi library misses an actually cool part of the midi spec called 'running status'. You can use this to optimize the midi stream a lot:
Some more info is here:
https://www.midikits.net/midi_analyser/running_status.htm

Should be in the handler though, otherwise the MIDI on the teensy is not compatible at all. Not sure if we can expect this?

PaulStoffregen, 2017-12-07:
USB MIDI uses a different format than serial MIDI.  Messages are packed into 32 bit fields, even if they would need only 2 or 3 bytes on serial MIDI.  Running status is not used with USB.

gritd, 2017-12-07:
hmm, well in that case it could be just a bug in the handling of the OP's program? Not sure if I understand his setup then.

PaulStoffregen, 2017-12-07:
Could also be a bug in this library.

The main hold-up is I don't have anything like that BopPad, and even if I did, I'm not a musician.  I need to find a reliable way to reproduce this problem, which is probably going to look like code on another Teensy which sends a rapid sequence of MIDI messages.

If anyone reading this issue wants me to work on a fix, coming up with a program I can run on another Teensy (without any user input) would be the way to help me move forward.  I can almost always fix problems when there's a way to reliably reproduce them....

PaulStoffregen, 2017-12-07:
Maybe this helps?  https://github.com/PaulStoffregen/USBHost_t36/commit/9e50acd012c2a50b41296cad239ea54ea7a4e80e

PaulStoffregen, 2017-12-13:
This fix is now available in Teensyduino 1.41-beta1.

https://forum.pjrc.com/threads/48409-Teensyduino-1-41-Beta-1

If you have hardware (and skill to play it fast enough) that can cause this issue to happen, please give 1.41-beta1 a try.  Does it fully resolve this problem?  Or do I need to look at other possible causes?

Dealerpriest, 2017-12-15:
I will look into this this afternoon! If it's not fixed I will try to provide a reproducible example.

Dealerpriest, 2017-12-15:
I have tried to reproduce the bug with the new version. But I couldn't. So everything points to that the made changes resolved the bug!
Yaaay!

closed 2017-12-15 by Dealerpriest

PaulStoffregen, 2017-12-16:
Thanks for the followup & testing!  :)