PaulStoffregen/USBHost_t36 #24: CP2104 doesn't appear to rx/tx data in the serial mode

ladyada, 2018-12-26:
tested two CP2104 boards/cables, they are detected but don't send or receive data. swapping in an FTDI cable works just fine. 
@KurtE dya remember if you tested CP210x? i could revert to an earlier commit if ya did :)

KurtE, 2018-12-26:
I believe I tested a few boards, but some could be different.  Which boards are not working?  Maybe need to pickup one of them...

ladyada, 2018-12-26:
hmm - thats odd - i tried a genuine CP2104 on a Metro Mini - https://www.adafruit.com/product/2590 and ran SerialTest from the latest master commit

KurtE, 2018-12-26:
Hi @ladyada, 

I believe I  only tested with one Silex CP210 board, I need to go figure out if I can find it.  Some of this code looks at VID:PID values to know which Serial board it is... Currently the only one setup for this is:
0x10c4:0xea60, which is in the library file serial.cpp. 

You might check to see what the pid/vid value is for your board. 

I might try to order one of yours or see if this is the one I tested with:
Like does this one work: https://www.adafruit.com/product/3309

May need to order one of the ones you mentioned: https://www.adafruit.com/product/2590

Need to see if Amazon carries it, as I can no longer order from you as I don't have a mailbox at my physical address as I don't want the mail delivered a half mile from my house, so I use a PMB in town...

But again you might try checking to see what the product and vendor ids are and edit that serial.cpp file table around line 58 and add it and see if it makes it work.

Kurt

ladyada, 2018-12-26:
Thanks i have a  https://www.adafruit.com/product/3309 and tried it and it also doesn't work. enumerates but no data goes out when i type into the serial console. 

The VID/PIDs all match and are correct - this is the `TestSerial` output:
```USB Host Testing - Serial
*** Device USERIAL1 10c4:ea60 - connected ***
  manufacturer: Silicon Labs
  product: CP2104 USB to UART Bridge Controller
Starting with baud: 115200
Serial Available
plain data
plain data
plain data
plain data
plain data
plain data
```
(i added a println right before `userial.write(ch);` to verify)

You mention "Silex", but these chips are SiLabs - just to verify, do you know what chip you tried? maybe it was a knockoff or clone?

PaulStoffregen, 2018-12-26:
I ordered the 3309 product.

KurtE, 2018-12-26:
Earlier today I ordered the two I mentioned (both the breakout board and the metro board) from Amazon.  I think the breakout board is scheduled to arrive Friday and the Metro on Monday. 

The one currently have is some generic one I think I either ordered on Amazon or E-Bay.

Markings on back say D-SUN, USB TO TTL, I will try it out again and also again try it when the Adafruit ones come.  As you mentioned, the one I have may not be an official one, or different model

The output from Serial test shows:
```
CP210X:  0x41, 0x11, 0, 0, 0 - reset port
*** Device USERIAL1 10c4:ea60 - connected ***
  manufacturer: Silicon Labs
  product: CP2102 USB to UART Bridge Controller
control callback (serial) F
control callback (serial) E
control callback (serial) C
control callback (serial) 8
control callback (serial) 0
```
I typed in some text, note, I have debug turned on to see what is going on:
```
Serial Available
txtimer
  TX data (21) 49 66 20 61 74 20 66 69 72 73 74 20 79 6F 75 20 64 6F 6E 74 0A 
rx token: 803F8100 transfer length: 64 len:1 - 49 0
rx: 49 
tx1:
Irx token: 3B8100 transfer length: 64 len:5 - 66 20
rx: 66 20 61 74 20 
f at rx token: 80348100 transfer length: 64 len:12 - 66 69
rx: 66 69 72 73 74 20 79 6F 75 20 64 6F 
first you dorx token: 3D8100 transfer length: 64 len:3 - 6E 74
rx: 6E 74 0A 
nt
```
With debug turned off, I have:
```

USB Host Testing - Serial
*** Device USERIAL1 10c4:ea60 - connected ***
  manufacturer: Silicon Labs
  product: CP2102 USB to UART Bridge Controller
Serial Available
I just typed in a line of text
```
So maybe 2102 is different than 2104 here.  Note: for this test I have the RX jumper to the TX pin of the  converter, such that everything that I type into the Arduino Monitor is sent out to the device which then sees the data and reads it back in and prints the data out.

ladyada, 2018-12-26:
ooh yeah 2102 is probably different than the cp2014 - but they have the same VID/PID? weird!

FrankBoesing, 2018-12-26:
According to the datasheets:
https://www.silabs.com/documents/public/data-sheets/CP2102-9.pdf
and
https://www.silabs.com/documents/public/data-sheets/cp2104.pdf

They have the same default VID/PID: 10C4h/EA60h If the OTP-ROM has not been customized.

KurtE, 2018-12-26:
Thanks Frank, 

Note: I just tried running the test again with one of my Saleae logic analyzers attached to the TX/RX pins, and verified that the default program was properly outputting at 115200.  I then recompiled for 9600 and the Autobaud setting on logic Analyzer said that the new baud as 9606, so looks like that part is working.  Again not sure what else I can check here until new hardware arrives.

ladyada, 2019-01-07:
hihi, just wondering if anyone else was able to replicate the bug (at least) - or if its something amiss with my setup!

KurtE, 2019-01-07:
Hi @ladyada (and @PaulStoffregen)

Yes - I am able to reproduce.  So far I have only tried with the Arduino board...  Which shows the problem. 

I know that some of the init code is failing.  I know that some of the chips had an issue where something like some register value gets reversed.  I have seen some tests that the Linux driver does to see if a board has the issue or not.  I did not put that level of support in yet and don't remember if the board I started off had it or not... My guess is one of these two boards has it and the other does not...

What I probably need to do is to run the setup code on a Linux box again and hopefully capture the USB startup sequence using my Saleae Logic Pro 8, and try to diagnose what packets and values are seen or not seen, ...   I Know earlier I would save away some reports and use something like grep and editors to figure out the data... But has been awhile

Problem is I am having too much fun trying to help out on some other projects, so I have looked any deeper yet.  Also don't have as much time due to 13 week old puppy

closed 2019-01-17 by PaulStoffregen

ladyada, 2019-01-17:
thanks - tested and works!

PR #25 (2019-01-17): Serial CP210x support for more boards
Updated some of the init code for the CP210x.

Before it worked with the one USB to serial adapter I had, but
we foudn it id did not work with Adafruit USB adaptor nor with their metro boards

This should fix #24 

Tested with my earlier CP210x board as well as the two adafruit boards