PaulStoffregen/USBHost_t36 #63: Is it possible to open a serial connection to a device?

MisterMelDev, 2021-06-15:
Hi there, thanks for this library.

I'm trying to connect to another microcontroller using a Teensy 4.0. I'd like to open a simple Serial connection to be able to read/write data. I can't find any examples which implement this (or I'm just understanding them incorrectly). Is that possible to do with this library? If so, how would I go about doing so?

ldoench, 2021-07-02:
I am having a similar problem: I have a HID device (which is a touchscreen), that i want to debug. Therefore I need some way to display the raw HID data, that is arriving at the USB host port. Maybe on the serial console of the host computer?
Anyway if there is any way to access the data on the USB host port of the Teensy, it would help me a lot.

FrankBoesing, 2021-07-02:
For questions, please use the forum: https://forum.pjrc.com