PaulStoffregen/USBHost_t36 #72: Original Novation Launchpad Mk1 not recognised?

nikscave, 2021-09-17:
hiya, would anyone have had experience with the original launchpad? USBHost_t36 works as expected with the launchpad mini, but the larger original launchpad show no signs of midi activity.
any pointers appreciated.
thanks
Nik

insolace, 2021-09-17:
The original Launchpad is not a class compliant MIDI device, it requires a driver from Novation to work and they only provide Win/MacOS. Someone would need to reverse engineer it to work with a Teensy/Arduino.

nikscave, 2021-09-18:
thanks for the info. much appreciated.
Does anyone know if any open source projects have started to look at that as an option? I can only find drivers relating to the Pro instance. hey ho.
thanks again.
Nik