PaulStoffregen/USBHost_t36 #91: Does not compile on VSCode

kopierreko, 2022-06-08:
Hi,

I'm testing your storage example and it's not compiling on VSCode with platformIO

`lib/USBHost/USBHost_t36.h:2095:66: error: expected class-name before '{' token
 class USBDrive : public USBDriver, public FsBlockDeviceInterface {`

VSCode 1.67.2 on macOS Monterey, platform : Teensy 4.16

PaulStoffregen, 2022-06-08:
Use the 1.56 released version.  Latest from github depends on new SdFat and core library.  If you want to use this, you'll need to update all those.  Or wait for platformio to update after we release 1.57.  Or use Arduino with 1.57-beta2.

https://forum.pjrc.com/threads/70409-Teensyduino-1-57-Beta-2

kopierreko, 2022-06-08:
Thanks, I copied SDFat from Teensyduino 1.57 beta 2 to my project in PlatformIO for the moment and it works 😊

closed 2022-06-08 by kopierreko