PaulStoffregen/USBHost_t36 #102: undefined reference to USBDrive::filesystem_assign_to_drive(USBFSBase*, bool)'

pr8x, 2022-12-04:
I am trying to run the "MSC List files" example using PlatformIO. This is my platformio.ini:

```ini
[env:teensy40]
platform = teensy
board = teensy40
framework = arduino
lib_deps = 
	https://github.com/PaulStoffregen/USBHost_t36/archive/refs/heads/master.zip
```

It should use the latest Teensyduino core 1.57. It compiles just fine, but fails to link:

> .pio\build\teensy40\lib411\libUSBHost_t36.a(MassStorageDriver.cpp.o): In function `USBFilesystem::begin(USBDrive*, bool, unsigned char)':
MassStorageDriver.cpp:(.text._ZN13USBFilesystem5beginEP8USBDrivebh+0x94): undefined reference to USBDrive::filesystem_assign_to_drive(USBFSBase*, bool)'

Any idea why this happens :/ ?

KurtE, 2022-12-05:
@wwatson4506, @mjs513,

Thanks, you are right that that method should have been removed from the header file.  We went from having each sketch have to scan the drive list to figure out which ones are new and then enumerate the partitions on them, to add the drives.  This is now handled for you.

Which Example are you using? 

If you are using ones from: https://github.com/wwatson4506/UsbMscFat
if you look at the readme file up on github:

It now starts off with:
```
UsbMscFat Library
NOTE: This Library is OUTDATED !!!
Please Use the examples in the latest USBHost_t36 stable library for the Teensy. Teensyduino 1.57
```

pr8x, 2022-12-05:
@KurtE I am using this example: https://github.com/PaulStoffregen/USBHost_t36/blob/master/examples/Storage/ListFiles/ListFiles.ino

closed 2022-12-11 by PaulStoffregen

pr8x, 2022-12-21:
@KurtE  I tried testing it. It now compiles (and links), but gets stuck during:

```c++
  while (!firstPartition) {
    myusb.Task();
  }
```

I am using a Teensy 4 and soldered pins to the Usb Host D+/D- pads on the backside. These are connected with a simple USB breakout board:

[image]

5V is provided to VBus via an external power supply.  The USB is formatted as FAT32. Am I missing something? Is there any verbose debug output I can enable? `DBGprint` doesn't seem to print anything.

KurtE, 2022-12-21:
The global debug setting is in USBHost_t36.h 
At about line 63: 
`//#define USBHOST_PRINT_DEBUG
`
Uncomment the line

Also MassStorageDriver.cpp has some additional debug output:
```
// Uncomment this to display function usage and sequencing.
//#define DBGprint 1
#ifdef DBGprint
#define DBGPrintf Serial.printf

```
I am assuming you have a common ground between this board, your 5v and the teensy

pr8x, 2022-12-21:
> I am assuming you have a common ground between this board, your 5v and the teensy

That's a good point. I connected grounds and enabled the debug flag.

I am now getting this output:

> Initializing usb host...
USB2 PLL running
 reset waited 6
USBHS_ASYNCLISTADDR = 0
USBHS_PERIODICLISTBASE = 2000B000
periodictable = 2000B000
Waiting for partition...
  begin reset
port change: 18001205
  port enabled
  end recovery
new_Device: 480 Mbit/sec
new_Pipe
enumeration:
enumeration:
enumeration:
Device Descriptor:
  12 01 10 02 00 00 00 40 64 85 00 10 00 11 01 02 03 01
    VendorID = 8564, ProductID = 1000, Version = 1100
    Class/Subclass/Protocol = 0 / 0 / 0
    Number of Configurations = 1
enumeration:
enumeration:
Manufacturer: JetFlash
enumeration:
Product: Mass Storage Device
ERROR Followup
    remain on followup list
    remain on followup list
    remain on followup list

PR #103 (2022-12-11): Update Storage Examples
Fixes #102

Several of the examples were not updated to work with the now automatic enumeration of partitions code.

I removed a few more pieces of the old manual code and updated the examples to reflect this.

I also updated the wave player example to play all of the wave files in the root directory of the storage instead of some fixed named ones.

@pr8x @mjs513 @wwatson4506 

Hopefully did not mess anything up. Pretty straight forward.