PaulStoffregen/USBHost_t36 #123: Odd behaviour with Behringer EDGE MIDI device seemingly not receiving messages

doctea, 2023-07-23:
Hello!

I've just got a Behringer EDGE (drum machine that supports USB-MIDI for a clock source), and am trying to add support for it to my custom USB MIDI host device that uses USBHost_t36.

I'm getting the following strange behaviour though:-
1. Device enumerates and identifies correctly, as with any other USB MIDI device I've tested.
2. However, when powering up the the EDGE and connecting it to the Teensy (via USB hubs), it does not seem to respond to any MIDI Start/Stop/etc messages.
3. Connecting the EDGE to my Windows PC for a moment - but not opening up any MIDI software - and then swapping the USB cable from my PC to the Teensy/USB hub, the EDGE starts responding as expected to the messages sent by the Teensy.

Its almost like connecting it to the PC is doing some sort of initialisation that is required for the EDGE to respond to MIDI.  Something that doesn't get reset when swapping the USB cable to the Teensy.

Any ideas what could cause something like this?  Or ways to tell what the Windows side is doing that the Teensy isn't, or things to attempt doing from the Teensy side that might make this work?

doctea, 2023-09-02:
To follow up on this, the behaviour reported in the original ticket was seen while using a fork of an older version of the library.

I've now updated my project to using the current head, and found that plugging the Edge in in the 'uninitialised' state either completely freezes the processing on the Teensy until the Edge is disconnected, or sometimes appears to crash it outright.

Doing the trick mentioned of 'initialising' the Edge by plugging it into my Windows PC first seems to work around this problem.

I enabled debugging in the USBHost_t36 library, but there doesn't appear to be any difference between 'uninitialised' and 'initialised'.  ('uninitialised' seems to show an extra port change/disconnect/connect cycle at the start of logs, but don't know if that is incidental?).

Debug info shown below in case it gives any clues what might be happening:-
```
port change: 10001803
    connect
port change: 1C001002 (this only shows on an 'uninitialised' connection)
    disconnect (this only shows on an 'uninitialised' connection)
port change: 10001803 (this only shows on an 'uninitialised' connection)
    connect (this only shows on an 'uninitialised' connection)
  begin reset
port change: 10001005
  port enabled
  end recovery
new_Device: 12 Mbit/sec
new_Pipe
enumeration:
enumeration:
enumeration:
Device Descriptor:
  12 01 00 02 00 00 00 40 97 13 5A 12 00 02 01 02 03 01 
    VendorID = 1397, ProductID = 125A, Version = 0200
    Class/Subclass/Protocol = 0 / 0 / 0
    Number of Configurations = 1
enumeration:
enumeration:
Manufacturer: Behringer
enumeration:
Product: EDGE
enumeration:
Serial Number: EDGE_V1.0.0
enumeration:
Config data length = 101
enumeration:
Configuration Descriptor:
  09 02 65 00 02 01 00 C0 00 
    NumInterfaces = 2
    ConfigurationValue = 1
  09 04 00 00 00 01 01 00 00 
    Interface = 0
    Number of endpoints = 0
    Class/Subclass/Protocol = 1 / 1 / 0
  09 24 01 00 01 09 00 01 01 
  09 04 01 00 02 01 03 00 00 
    Interface = 1
    Number of endpoints = 2
    Class/Subclass/Protocol = 1 / 3 / 0
  07 24 01 00 01 41 00 
  06 24 02 01 01 00 
  06 24 02 02 02 00 
  09 24 03 01 03 01 02 01 00 
  09 24 03 02 04 01 01 01 00 
  09 05 02 02 40 00 00 00 00 
    Endpoint = 2 OUT
    Type = Bulk
    Max Size = 64
    Polling Interval = 0
  05 25 01 01 01 
  09 05 81 02 40 00 00 00 00 
    Endpoint = 1 IN
    Type = Bulk
    Max Size = 64
    Polling Interval = 0
  05 25 01 01 03 
[... 103 lines cut ...]
type: 36, len: 9
    MIDI OUT Jack (ignored)
type: 36, len: 9
    MIDI OUT Jack (ignored)
type: 5, len: 9
    MIDI Endpoint: 2
      tx_size = 64
type: 37, len: 5
    MIDI Endpoint Jack Association (ignored)
type: 5, len: 9
    MIDI Endpoint: 81
      rx_size = 64
type: 37, len: 5
    MIDI Endpoint Jack Association (ignored)
new_Pipe
new_Pipe
Descriptor 36 =  ???
Descriptor 36 =  ???
Descriptor 36 =  ???
Descriptor 36 =  ???
Descriptor 36 =  ???
Descriptor 5 = ENDPOINT
Descriptor 37 =  ???
Descriptor 5 = ENDPOINT
Descriptor 37 =  ???```

pierremestrez1976, 2024-07-03:
Hello,

I encounter the same behavior with 3 different Behringer TD3 synths.   When connected to PC/DAW first, their USB port behave normally (like Arturia Beatstep, Elektron Model Samples, Toraiz Squid etc...) but when the TD3 are powered on without having been connected to a PC/DAW, they stuck when entering the loop().

The only message I send in my loop is  USBDevicesList[port]->sendRealTime(0xF8, 0);  
For test, it is sent as tick for 24 ppqn at 120 BPM.

It seems this message affect MIDIDeviceBase::write_packed and/or MIDIDeviceBase::timer_event because I see different values;

1. When the TD3 has been connected to PC/DAW:

tx1 = 1
tx1_count = 1
tx2 = 0
tx2_count = 0

tx1 = 1
tx1_count = 1
tx2 = 0
tx2_count = 0

2. When the TD3 has NOT been connected to PC/DAW:

tx1 = 1
tx1_count = 1
tx2 = 0
tx2_count = 0

tx1 = 16
tx1_count = 16
tx2 = 1
tx2_count = 1

This combination of variable values tx1 = 16 and tx2 = 1 seems to freeze the Teensy...

Any idea ?

doctea, 2024-07-14:
For some extra context around this problem -- @pierremestrez1976 reported to me privately that when using the standard release USBHost_t36 library, the Teensy freezes/blocks when the TD3 is connected; however, with the patched library (using [this branch](https://github.com/PaulStoffregen/USBHost_t36/compare/master...doctea:USBHost_t36:master)), no freeze happens but all MIDI messages seem to be ignored, unless the offending device is first connected to a PC.

I'll also mention that I reported this to Behringer/Music Tribe, and they said they could not find a problem and that their devices are MIDI Class Compliant, and so should work without problem.